waveshareteam/LCD-show #3: How to power off 3.5inch LCD when unused?

liluom, 2019-01-17:
Hi. Can i power off it (3.5inch RPi LCD (A)) for power saver by python? Thank you!

waveshare, 2019-01-21:
You can power off it by press the power button.

pistonov, 2019-04-06:
And how?

SSQ66648, 2020-07-18:
i am currently experimenting with https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A) on a RPi 3B+.
the driver seems to work and rotating screen (./LCD35-show 720) etc works. 
But how do we turn off the backlight after shutdown?
I have looked through the wiki as you mentioned but there is no mention of the backlight on that page and I cannot find anything else that seems relevent?
could you please provide instructions on howw to do this or even a link to the wiki instructions?
many thanks

jytou, 2020-09-03:
I have the same question here. How to programmatically switch off the screen? What about changing the brightness (what the small button on the back does). Thanks.

eschava, 2020-10-03:
Will instructions from https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(C) work for LCD (A) ?

Leshauts, 2023-10-27:
This doc shows how to install and adjust brightness, but it doesn't turn off the backlight. Also, even with this soft installed, when the display goes to idle, the backlight stays on...

sohail-bakhtiari, 2024-11-19:
> Will instructions from [https://www.waveshare.com/wiki/3.5inch_ (9 cm)˜RPi_LCD_(C)](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(C)) work for LCD (A) ?

The pads indicated there does not exist for LCD (A). Neither does that small transistor.