waveshareteam/LCD-show #12: Wrong CMakeCache.txt directory

vadimk2016, 2019-05-12:
After it runs the sudo ./LCD5-show command, it throws an error:
`CMake Error: The current CMakeCache.txt directory /home/pi/downloads/LCD-show/rpi-fbcp/build/CMakeCache.txt is different than the directory /home/pi/LCD-show/rpi-fbcp/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt CMake Error: The source "/home/pi/downloads/LCD-show/rpi-fbcp/CMakeLists.txt" does not match the source "/home/pi/LCD-show/rpi-fbcp/CMakeLists.txt" used to generate cache. Re-run cmake with a different source directory. CMake Error: The source directory "/home/pi/LCD-show/rpi-fbcp" does not exist. Specify --help for usage, or press the help button on the CMake GUI. Makefile:176: recipe for target 'cmake_check_build_system' failed make: *** [cmake_check_build_system] Error 1 LCD configrue 0`

LongBeachHXC, 2019-08-18:
nothing on this one? I'm getting this one too.

0x0056, 2019-10-24:
You have the LCD-show folder in downloads folder /home/pi/downloads/LCD-show/ must be on /home/pi/LCD-show/ solved moving the directory to home path

Ventusfahrer, 2019-12-08:
who the hell is using raspbian default user pi for doing anything else than booting the os creating an individual admin account?

For me it is best security practice to not use pi as user.

ericliang, 2020-04-18:
Same issue. Anyone know something about this package?

JimLynchCodes, 2021-02-01:
I ran into this issue after running a c++ program on a new computer than the one I had originally ran it on.

The program used a `CMakeLists.txt` to do the build initially and then created a `CMakeCache.txt` file.

When I deleted the `CMakeCache.txt` and ran cmake again it worked as expected.  👍

babatunde-ayoade, 2022-12-11:
> I ran into this issue after running a c++ program on a new computer than the one I had originally ran it on.
> 
> The program used a `CMakeLists.txt` to do the build initially and then created a `CMakeCache.txt` file.
> 
> When I deleted the `CMakeCache.txt` and ran cmake again it worked as expected. 👍

Thank you...THIS WORKED...... I have been stuck at this spot for 3 days now.

mcanvar, 2024-01-20:
> who the hell is using raspbian default user pi for doing anything else than booting the os creating an individual admin account?
> 
> For me it is best security practice to not use pi as user.

This should definitely fixed by the team. Thank you