waveshareteam/LCD-show #17: Absolutely abysmal scripts that are highly likely to brick your pi

True-Demon, 2020-01-10:
I'll keep this brief... your scripts have _multiple_ points of failure that should result in an abort or at the very least provide warnings, prompts, and debugging output to help narrow down the causes. These scripts have no such protections. For instance, cmake failed for me, resulting in no driver being built. Then it promptly overwrote critical files like my cmdline.txt and config.txt based on teh assumption that cmake would succeed and promptly rebooted without asking. 

Needless to say, your script bricked my pi and obliterated a week's worth of work I had been doing to prepare a custom image.

You use hard-coded paths, and make a ton of assumptions based solely on the Rasbian image, overwrite files without backing them up, and make dozens of extremely destructive changes that universally result in a kernel panic if anything goes wrong. Can't you at _least_ do some error handling FFS?

phixion, 2020-01-21:
[Yeah](https://github.com/waveshare/LCD-show/issues/16) ...noone reads this it seems. There's PRs from a year ago. I was checking their github since I wasn't willing to insert a cd-rom from china into anything that can read it, when I stumbled upon their "scripts"

razgithub, 2020-04-27:
Agree. These are pretty horrible!