waveshareteam/LCD-show #22: Kernel 5.4 support

autr, 2020-06-24:
Hello, I tested a 3.2 LCD with kernel 5.4 on Raspberry Pi OS, but isn't working yet - would be great to get this up and running as 5.4 adds libcamera and IMX477 support! Thank you

FinlandApollo, 2020-07-23:
I'm eager to know as my 3.5" LCD with 5.4. kernel is just blank screen.

autr, 2020-07-23:
Check the latest waveshare-dtoverlays;

https://github.com/swkim01/waveshare-dtoverlays

Where you can compile the latest device tree:

```
dtc -O dtb -o <output_filename>.dtb <input_filename>.dts
```

So for for 3.5" (version 3);

```
git clone https://github.com/swkim01/waveshare-dtoverlays
cd waveshare-dtoverlays
dtc -O dtb -o waveshare35c.dtb waveshare35c.dts
cp waveshare35c.dtb ../location/of/LCD-show/waveshare35c.dtb
# not follow LCD-show instructions...
```

It would make sense that this repo contains the dts as a submodule, and compiles the binary during the install script, but then, this repo is pretty sketchy.

FinlandApollo, 2020-07-23:
> Check the latest waveshare-dtoverlays;
>
> https://github.com/swkim01/waveshare-dtoverlays
>
> Where you can compile the latest device tree:
>
> ```
> dtc -O dtb -o <output_filename>.dtb <input_filename>.dts [...]

Does this work with 35b-v2 if I just replace that waveshare35c.dtb and .dts files with those waveshare35b-v2.dts file? 

The waveshare35a works but it makes colors inverted.

autr, 2020-07-24:
Sorry, typo, meant to say: _now_ follow LCD-show instructions

It might be better if I explain how I got there. While using a 3.2" V3 (ie. 32C) I found this commit and issue:

https://github.com/swkim01/waveshare-dtoverlays/commit/f0bdf7200c5822d73d2e013712c98903bfd96ca8
https://github.com/swkim01/waveshare-dtoverlays/issues/24

This is for 35A (ie. 3.5" V1), so I just replicated the change onto 32C dtoverlay:

https://github.com/swkim01/waveshare-dtoverlays/commit/0f2e3aca3e015c61a08229a0598eee717da73fa7

So for your LCD, you should either be able to use the 35A dtoverlay, or make the same change to 35B / C, then recompile.

autr, 2020-07-24:
From what I can glean from the discussions here, the fbtft driver switched to using "gpio descriptors", so its an off-by-one error:

https://github.com/torvalds/linux/commit/c440eee1a7a1d0f2d5fc2ee6049e4a05da540f01
https://github.com/notro/fbtft/issues/546

Zaryob, 2020-07-25:
I'm also living same issue. Whenever you publish 64bit OS without that support we wont be able to use it.

Apteryks, 2022-12-27:
That's long been merged into this repo and the waveshare35b display (and others) work by simply adding `dtoverlay=waveshare35b` to your config.txt (tested using kernel8.img, the official 64-bit kernel used on raspbian bullseye 11).

JimJ92120, 2023-12-17:
got it running with kernel `6.1` on `bullseye` following [**waveshare wiki instructions**] (https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(C))  
(**C** version)  

then add updating `/usr/share/X11/xorg.conf.d/99-fbdev.conf` as mentioned here:  
https://forums.raspberrypi.com/viewtopic.php?t=161512#p1048180