waveshareteam/LCD-show #26: License

fkromer, 2020-07-06:
Could you provide a license file pls?

mainejunction, 2020-07-06:
Do not understand... which license... Sent via the Samsung Galaxy Note10+, an AT&T 5G Evolution capable smartphone

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/waveshare/LCD-show/issues/26",
"url": "https://github.com/waveshare/LCD-show/issues/26",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

fkromer, 2020-07-06:
Customers which buy a screen from you and want to use the drivers in this repository in a commercial product need info about how Waveshare licenses the driver software. On GitHub a license is most often provided as license file (something like `LICENSE.md`, `LICENSE.txt`). Common license choices are e.g. [Apache-2.0](https://choosealicense.com/licenses/apache-2.0/) or [MIT](https://choosealicense.com/licenses/mit/) .

McDisaster, 2020-11-12:
It would be great if the repository is proper licensed (even better if it would be an open source license :smile: ).

When nothing explicitly is specified all rights are reserved by the copyright holder and it is **not** open source. This means you cannot redistribute anything (code/binaries) with for example a yocto build image.