waveshareteam/LCD-show #28: Raspberry Pi OS (Raspbian) 5.x module error

Zaryob, 2020-07-26:
Given module for every device has been problemmatic with 5.x linux version which would be default for Raspbian. 

Have you planning re-share TFT compiled drivers for that update.

Zaryob, 2020-07-26:
I already opened an issue on [Raspberry Pi Github](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/77) you can click to see.

2nt0, 2020-08-06:
Anyone else having this issue, solution for me (on RPi 4B) was to downgrade to firmware 4.19 `rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2` from https://github.com/goodtft/LCD-show/issues/223#issuecomment-661279921. Still not a great solution but it works for now.

Zaryob, 2020-08-09:
> Anyone else having this issue, solution for me (on RPi 4B) was to downgrade to firmware 4.19 `rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2` from [goodtft/LCD-show#223 (comment)](https://github.com/goodtft/LCD-show/issues/223#issuecomment-661279921). Still not a great solution but it works for now.

Hi. I fixed issue on my repository. https://github.com/Zaryob/LCD-show  at commit 9513c163163effa3fc5822cafa324b41ae205ea4 and also published as pr for this repository #30

closed 2020-12-14 by Zaryob

Apteryks, 2022-12-22:
Why is this closed?  Nothing has been merged to this repo, from what I can see?