waveshareteam/LCD-show #29: Touch input works yet display output is still on HDMI

charga600, 2020-08-06:
Recently started working with Pi 3 A+ and 3.2" RPi LCD (B). Installed Raspberry Pi OS using the imager from windows, updated once running on Pi and then cloned the git. After restart and switching to LCD display using:
`cd LCD-show/` and `./LCD32-show`
After reboot screen resolution changes and touchscreen becomes active, yet display still only outputs to HDMI connection.

2nt0, 2020-08-06:
downgrading kernel to 4.19 is a temporary fix (tested on 4B)
`rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2`

charga600, 2020-08-06:
> downgrading kernel to 4.19 is a temporary fix (tested on 4B)
> `rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2`

Works a charm, thank you very much.

closed 2020-08-06 by charga600