waveshareteam/LCD-show #31: ARM64 support

ChaimEvans, 2020-08-29:
In arm64, it can be installed successfully, but after that, the system can't start, and HDMI outputs a lot of words that can't be read clearly.I can't control it. It seems that I can only flash the system again.
(raspberry 4b, system:2020-08-20-raspios-buster-arm64)

innns, 2020-09-27:
After I run `./LCD35-show` 
Terminal shows
```
CMake Error: The current CMakeCache.txt directory /home/pi/tmp/LCD-show/rpi-fbcp/build/CMakeCache.txt is different than the directory /home/pi/LCD-show/rpi-fbcp/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
CMake Error: The source "/home/pi/tmp/LCD-show/rpi-fbcp/CMakeLists.txt" does not match the source "/home/pi/LCD-show/rpi-fbcp/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
CMake Error: The source directory "/home/pi/LCD-show/rpi-fbcp" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** [Makefile:176: cmake_check_build_system] Error 1
Selecting previously unselected package xinput-calibrator:armhf.
(Reading database ... 93052 files and directories currently installed.)
Preparing to unpack .../xinput-calibrator_0.7.5-1_armhf.deb ...
Unpacking xinput-calibrator:armhf (0.7.5-1) ...
dpkg: dependency problems prevent configuration of xinput-calibrator:armhf:
 xinput-calibrator:armhf depends on libstdc++6 (>= 4.6).
 xinput-calibrator:armhf depends on libx11-6.
 xinput-calibrator:armhf depends on libxext6.
 xinput-calibrator:armhf depends on libxi6 (>= 2:1.2.0).

dpkg: error processing package xinput-calibrator:armhf (--install):
 dependency problems - leaving unconfigured
Processing triggers for gnome-menus (3.31.4-3) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 xinput-calibrator:armhf
LCD configure 0
Connection to 192.168.3.133 closed by remote host.
Connection to 192.168.3.133 closed.
```
And I can't even ssh to the Pi
(raspberry 4b, system:2020-08-20-raspios-buster-arm64)

ChaimEvans, 2020-10-01:
At first I was in the same situation as you. In order to solve this problem, I deleted the code "**reboot**" in "**LCD35-show**" to make sure system doesn't restart automatically. Then solve various problems, such as dependency and compilation errors.Finaly manual restart until no error is reported, but still unable to start system.

lepatrick, 2021-01-03:
Should be compiled for arm64. @ChaimEvans can you please share your changes?

nofacer, 2021-02-09:
I tried on Ubuntu Server 20.04.02. 
* Changed all "pi" in code to "ubuntu".
* Install those manually:
  sudo apt-get install libraspberrypi-dev
  sudo apt-get install xinput-calibrator
* Comment this line
  # sudo dpkg -i -B ./xinput-calibrator_0.7.5-1_armhf.deb

Run sudo ./LCD35C-show, no errors. But the screen doesn't show anything(just change from gray to black after reboot)

Apteryks, 2022-12-22:
Someone apparently made it work for waveshare-35A: https://github.com/marcin-chwedczuk/waveshare-35A-raspberry-pi-64-driver

closed 2022-12-23 by ChaimEvans