waveshareteam/LCD-show #33: LCD white after installing drivers from git

entersb, 2020-09-21:
I've installed latest raspberry pi OS and followed instructions from wiki to install driver.

My LCD screen is white but touch options are working. 

Only thing regarding display that changes is that my HDMI device displays in resolution intended for LCD screen.

LCD screen is 3.5inch RPi LCD (A) V3.

svadivazagan, 2020-11-29:
> I've installed latest raspberry pi OS and followed instructions from wiki to install driver.
>
> My LCD screen is white but touch options are working.
>
> Only thing regarding display that changes is that [...]

I am using the waveshare 7 inch (H) type having the same issue but after twenty to twenty five minutes the screen comes to normal gradually.

fouadfawas, 2025-06-26:
> I am using the waveshare 7 inch (H) type having the same issue but after twenty to twenty five minutes the screen comes to normal gradually.

Can you use the waveshare 3.5 inch display on ubuntu and also use hdmi output at the same time?