waveshareteam/LCD-show #36: Can we get an update?

ArcAIN6, 2021-01-14:
Can we get an update here? 2 years is literally a LIFETIME. 

There are also a TON of highly destructive, amateurish issues with this script-kiddie approach here. 

You overwrite files, and don't make backups (Really?? REALLY???!!!!)
You "ASSUME" that everyone is running Raspbian... NOPE. 
You "ASSUME" that everything is where YOU think it should be. 

There is literally NOTHING here worth salvaging.. 

Honestly.. it's like you got it to work once, in a specific case situation, and then regurgitated it into a ton of scripts.. 

i'll never buy another waveshare product again.. .sorry, but not sorry, this is a bare minimum approach. 

If you can't support your products, stop making products, you don't deserve the income, ,go back to producing one off electronics for yourself, because as a business model, this is doomed to failure.

xversial, 2021-02-26:
I'm disappointed to see the lack of response from the manufacturer.

ArcAIN6, 2021-02-26:
> 
> 
> I'm disappointed to see the lack of response from the manufacturer.

It's ridiculous right?
Also.. cute avatar <3

xversial, 2021-03-04:
> > I'm disappointed to see the lack of response from the manufacturer.
> 
> It's ridiculous right?
> Also.. cute avatar <3

Thanks ;3

snoweuph, 2021-07-11:
yay caant use spi after  installing  touch drivers, but i  have no hope  for help with  the manufactor  to help  me  :)

dabulla, 2021-08-08:
Give them a bad review on amazon, maybe they will start helping then.
I bricked my 5th OS installation with these stupid scripts now.
I think manjaro is not supported, 64bit is not supported, wayland is not supported... but the script can successfully brick everything.