waveshareteam/LCD-show #38: Using TFT 3.5'' with this library, disables SPI completely

mmzeynalli, 2021-02-05:
Good day,
Today I struggled with a problem, where I use a TFT display with Raspberry PI 3B+. In addition to this, I use SX1276 radio module, which also works with SPI protocol. Although I can use both of them separately, I could not use them both at the same time. The issue was that after the call of `LCD35-show` command is executed, after reboot, the SPI is disabled, and no `spidev` device can be found in `/dev/` directory. Manually turning on SPI from raspi-config does not solve this issue. Is there any solution or at least any workaround to this problem?

damik422, 2021-09-13:
Same problem here, have you figured something out?

mmzeynalli, 2021-09-14:
To be honest no, I had to change screen and use another one

damik422, 2021-09-14:
[image]
solved it by commenting this line (/boot/config.txt), as i have a hdmi display the touch still works

mmzeynalli, 2021-09-14:
And can you use another SPI device?

damik422, 2021-09-14:
I was able to use the ADXL345 accelerometer connected to spi0 bus of rpi after commenting out this line, yes, previously it didn't work. This line is what seems to disable SPI

mmzeynalli, 2021-09-14:
Perfect. Thank you!