waveshareteam/LCD-show #43: Does not see the Bluetooth Adapter after implementing.

KaraCemal, 2021-04-10:
Hi,
I am using Raspberry Pi 3B+ and also using Bluepy(python library) for Bluetooth Low Energy signal detecting.
After switching to the LCD screen (ie. sudo ./LCD32-show) the Bluetooth doesn't seem to be working.
In bluetoothctl, 'Scan on' and 'Power on' commands return "No Bluetooth Adapter Found".
Re-switching or rebooting the machine does not solve the problem, Raspberry does not see the Adapter after that point.
Only solution on my side is re-imaging the RP, but I have to implement the screen.
Do you have any ideas for my specific problem?
Thanks in advance!

macatoto, 2021-04-13:
Hi @KaraCemal , I had the same problem, took me a while, but finally I found [this](https://github.com/goodtft/LCD-show/issues/278) post which did the trick for me:

edit /boot/cmdline.txt and replace 

<code>console=ttyAMA0,115200</code>

with 

<code>console=serial0,115200</code>

KaraCemal, 2021-04-13:
Hi @macatoto , thanks for the reply. 
I also just found the trick last night and it solved problem directly. The issue is somehow related with latest Raspberry Pi OS update and SPI connection I think. 

However, this Icon Lost occurs every time switching from HDMI to LCD and I wonder if there is a permanent solution for that.

The related forum post is also here:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&hilit=LCD%203.5%20bluetooth&p=1836645&t=306699&sid=ccbf7f23c54617d1e1bb885052556c84

neicureuil, 2021-05-21:
> 
> 
> Hi @KaraCemal , I had the same problem, took me a while, but finally I found [this](https://github.com/goodtft/LCD-show/issues/278) post which did the trick for me:
> 
> edit /boot/cmdline.txt and replace
> 
> `console=ttyAMA0,115200`
> 
> with
> 
> `console=serial0,115200`

I have the same problem, and your answer save me. Thanks
For a automaticly fix the issue each time, you can just make this change in the `cmdline.txt` file which is in the LCD-Show folder