waveshareteam/LCD-show #45: spi not  working after touch-driver installation

snoweuph, 2021-07-11:
I have this screen: https://www.waveshare.com/wiki/5inch_HDMI_LCD
before i installed the touch driver i could use SPI in python without problems, now after installing the driver i get the error:
```
      spi.open (0.0)
FileNotFoundError: [Errno 2] No such files or directory 
```

snoweuph, 2021-07-11:
If I commment out this line  in the config.txt everything is  working fine  again, without touch
```
dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900
```

damik422, 2021-09-13:
I have the same problem, is it fixed yet?

snoweuph, 2023-06-11:
> 

Waveshare is the Typical Chinese Company that has nice hardware but suck at software