waveshareteam/LCD-show #52: no boot screen after install raspberry 3b+

Blac71, 2021-12-27:
install driver after lcd reboot only the cursor flashes i don't see the boot process
[image]

atabou, 2022-10-24:
Same problem here with raspberry pi 4 and waveshare lcd 3.5 inch B.

- Booting headless leads to this state.

- Booting with an hdmi cable connected to a monitor and with the lcd plugged, the lcd boots to the same screen as above. The monitor on hdmi1 shows a 480x320 desktop environment. The touch functionality works but the effect takes place on the monitor.

- Now removing the lcd screen still leads the rpi to think it is still connected to the lcd screen.

Somehow, it seems that the driver redirects all output to hdmi1 regardles of how things are connected.

1DevilishAngel, 2023-04-13:
I had the same problem and managed to fix by installing the lite version of the driver