waveshareteam/LCD-show #53: LCD35-Show touch screen not working

idzuwan, 2022-02-10:
Installed using LCD35-Show lite, display are fine but touch screen does not seem to work at all, any howtos for troubleshooting?

clayy503, 2022-02-19:
Same with the LCD 3.2 (b) v4. Installed latest version of Raspian desktop on my 2w and only the screen display works after installing drivers, no touch.

mcreality23, 2022-02-20:
Same here with LCD 3.5 B v2. Latest Raspbian Desktop on Pi400.

mrt2407, 2022-04-06:
Same with LCD 10.1. Worked for a while but every now and then it isn't working.

armandoisaac, 2022-05-10:
Has anyone been able to make the touch work? I bought the screen to use it with as OctoDash but is worthless without the touch functionality.

JimJ92120, 2023-12-17:
got it running on **Pi 4B** with kernel `6.1` on `bullseye` and `bookworm` (**Raspberry PI OS 64-bit version**) following [**waveshare wiki instructions**] (https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(C)#Install_the_touch_driver)  
(example for **C** version)   

also assuming `x-org` is used (and not `wayland` on `bookworm`)  

following must be run with `root` privilege:  
```sh
./LCD35C-show

# now
sudo  ./LCD35C-show
```

then updating (or creating) `/usr/share/X11/xorg.conf.d/99-fbdev.conf` as mentioned here:  
https://forums.raspberrypi.com/viewtopic.php?t=161512#p1048180  

```conf
...
  # output to LCD screen
  Option "fbdev" "/dev/fb1"

  # or replace with this to output to HDMI instead
  Option "fbdev" "/dev/fb0"
...
```