waveshareteam/LCD-show #55: LCD-35C-show not compatible with Raspberry Pi 4 Kali Linux 64 bit

Aethernite, 2022-04-20:
After installing the 64-bit Kali Linux on my Raspberry Pi 4 the LCD show script doesn't seem to work. It fails to pull dependencies and reboots the RPI. The result from that is a white screen and HDMI connector not working (because the script disables it). To go around that issue I had to use the 32-bit Kali Linux image.

dazbaker, 2022-04-26:
same here with the same thing ive just upgraded to the 64 all i get is a blank screen?

LPDaVinci, 2022-06-14:
i have the problem that when using the newest kali arm version i only get the terminal on boot and lightdm dont works anymore. when using the lcdwiki mhs35-show it works and is fast... the 35-show from lcdwiki is laggy...

Mifaxav, 2023-05-06:
it will only work with 32 bit os's it seems

JimJ92120, 2023-12-17:
got it running on **Pi 4B** with kernel `6.1` on `bullseye` and `bookworm` (**Raspberry PI OS 64-bit version**) following [**waveshare wiki instructions**] (https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(C)#Install_the_touch_driver)  
(example for **C** version)   

also assuming `x-org` is used (and not `wayland` on `bookworm`)  

following must be run with `root` privilege:  
```sh
./LCD35C-show

# now
sudo  ./LCD35C-show
```

then updating (or creating) `/usr/share/X11/xorg.conf.d/99-fbdev.conf` as mentioned here:  
https://forums.raspberrypi.com/viewtopic.php?t=161512#p1048180  

```conf
...
  # output to LCD screen
  Option "fbdev" "/dev/fb1"

  # or replace with this to output to HDMI instead
  Option "fbdev" "/dev/fb0"
...
```

Mifaxav, 2024-01-09:
cool..! ill try this out when i need the screen again