waveshareteam/LCD-show #56: Please fix LCD35C-show. Has not been updated in 4 years and doesn't work.

Babyyoda777, 2022-12-06:
goodtft/LCD-show does work but unfortunately they don't have the LCD3.5C variant and the touch doesn't work. The official driver doesn't work at all. Please fix products which you sell. no point selling it if you are not going to actually support it and maintain it.

snio89, 2023-04-12:
`LCD35-show` of `goodtft/LCD-show` does work for my 3.5C, both for the display and touch.

JimJ92120, 2023-12-17:
got it running on **Pi 4B** with kernel `6.1` on `bullseye` and `bookworm` (**Raspberry PI OS 64-bit version**) following [**waveshare wiki instructions**] (https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(C)#Install_the_touch_driver)  
(example for **C** version)   

also assuming `x-org` is used (and not `wayland` on `bookworm`)  

following must be run with `root` privilege:  
```sh
./LCD35C-show

# now
sudo  ./LCD35C-show
```

then updating (or creating) `/usr/share/X11/xorg.conf.d/99-fbdev.conf` as mentioned here:  
https://forums.raspberrypi.com/viewtopic.php?t=161512#p1048180  

```conf
...
  # output to LCD screen
  Option "fbdev" "/dev/fb1"

  # or replace with this to output to HDMI instead
  Option "fbdev" "/dev/fb0"
...
```