waveshareteam/LCD-show #61: 3.5inch RPi LCD (B) not working on Raspberry Pi OS Lite (bookworm)

project-owner, 2023-10-18:
There were no errors when I ran the script:
```
git clone https://github.com/waveshare/LCD-show.git
cd LCD-show/
./LCD35B-show lite
```
I can see the boot messages on the screen and  the touchscreen calibration worked fine.
But the display doesn't show any image when it's working either with Pygame 2 and framebuffer or just framebuffer 'fbi' utility.
Is there any way to make it working?
Thanks!

Toh-git, 2023-10-23:
Hello, 
 I have BookWorm and LCD35-show didnot work as well. Any solution on this problem?

project-owner, 2023-10-25:
I've done more experiments and can say for sure that this is Waveshare display problem: https://forums.raspberrypi.com/viewtopic.php?t=358144#p2149219
I'm not sure if that's hardware or software issue. Hopefully software issue which can be fixed.

JimJ92120, 2023-12-18:
@project-owner can you try that maybe? https://github.com/waveshare/LCD-show/issues/58#issuecomment-1859084420  
=> i assume same could apply for other versions (A et B)

worked for me on with `bookworm` and `bullseye` for `Rasberry Pi OS`  
it won't likely work for `non-Raspbian` based OS - e.g `Ubuntu` - as using a different kernel (so likely more changes)  

also must run through `xorg` (repo doesn't not setup for e.g `wayland`)

project-owner, 2024-01-11:
I need it on OS Lite - the OS without desktop.

JimJ92120, 2024-01-11:
@project-owner both works are working on my side. 
additionally `LCD35B` model has 2 version models. are you correctly using the `v1` or v2` and matching script?

project-owner, 2024-01-11:
Yes, the terminal shows up on a screen. But this is not enough for me. I need to draw on screen using Pygame 2 llibrary. That doesn't work. Here is the simple Python script which draws a line on a screen. This script doesn't work.
```
import os
import pygame
import time
os.environ["SDL_FBDEV"] = "/dev/fb0"
pygame.init()
display = pygame.display.set_mode((480, 320))
print(pygame.display.get_driver())
display.fill((0,0,0))
pygame.display.update()
pygame.draw.line(display, (30,80,200),[100,100],[200,200],2);
pygame.display.update()
time.sleep(5)
```
I have v1 display. So, I use this command:
```
./LCD35B-show lite
```

JimJ92120, 2024-01-11:
@project-owner I believe it has nothing to do with driver and this repo then.
Driver is working as you're able to render what the OS is supposed to.

A server **do not render GUI** by default and by design except the usual **terminal**, unless such GUI rendering is explicitly configured and declared.
That's why there are server and desktop versions.

So this would point to your system being misconfigured and the need to set up such rendering and forward it to your screen.

You may search for "kiosk mode" OS's - `rpi-imager` does have some in images list.

vchrizz, 2024-06-27:
It still works fine, since Raspbian bookworm the files config.txt and cmdline.txt moved from `/boot/` to `/boot/firmware/` .
So I did following **before** I run ./LCD-show installation:
```
mv /boot/config.txt /boot/config.txt.old
mv /boot/cmdline.txt /boot/cmdline.txt.old
ln -s /boot/firmware/config.txt /boot/config.txt
ln -s /boot/firmware/cmdline.txt /boot/cmdline.txt
```

project-owner, 2024-06-28:
Which OS do you use, lite or desktop version? I need it working on a Lite version (no X, no Wayland). Can you use Pygame?