waveshareteam/LCD-show #64: Not working on raspberry pi 5

Giuseppe7887, 2024-01-24:
raspberry pi 5, display lcd rpi c, your driver show35C-show, your config file config-35c.txt, system not updated, can i know what am i doing wrong?

hannadrehman, 2024-01-26:
I have pi5 and 3.5(A). It simply kills the display. No output

mahdiabbasinourabadi, 2024-05-04:
PI5 + 3.5B LCD TOUCH , NO OUTPUT !!

EngineeringOV, 2024-06-15:
Same here, would REALLY love a fix.

vchrizz, 2024-06-27:
Do you use Raspbian bookworm? If yes:
It still works fine, since Raspbian bookworm the files config.txt and cmdline.txt moved from `/boot/` to `/boot/firmware/` .
So I did following **before** I run ./LCD-show installation:
```
mv /boot/config.txt /boot/config.txt.old
mv /boot/cmdline.txt /boot/cmdline.txt.old
ln -s /boot/firmware/config.txt /boot/config.txt
ln -s /boot/firmware/cmdline.txt /boot/cmdline.txt
```

Giuseppe7887, 2024-06-28:
didn't work yet

EngineeringOV, 2024-06-30:
The problem isn't the configs, it's that FBCP is no longer supported.

madscientist42, 2025-01-20:
DispmanX was deprecated to the point that it wasn't offered as a default feature on Bookworm and DispmanX isn't offered on a Pi5.

The Pi Foundation and the Kernel community moved to supporting KMS/DRM panel support back around this time LAST year (Around the time of the ticket here...)   [https://forums.raspberrypi.com/viewtopic.php?p=2190025](https://forums.raspberrypi.com/viewtopic.php?p=2190025)  

KMS/DRM support has been provided by many of the mainline players...and in the RaspberryPI kernels, etc.  Yet Waveshare?  Conspicuously missing.  Used to think highly of them...now?  I've got a batch of HW, that they're still selling out of their store and out of Amazon, that absolutely does not work...and in a manner that they know or should've known (Instead of that song and dance we got in this thread...)

Step up to the plate and support your hardware you're STILL SELLING or quit selling it and we'll figure out how to fix your mess...but in that case you'll be put to the least favored supplier status on that note.

tscholze, 2025-05-05:
So in short, this will not work on a now years old Raspberry Pi 5? So my setup is for the bin? Hm. Don'T think it's waveshare fault but the seller should clearly mark if this is Pi5 compatible or not.

Giuseppe7887, 2025-05-05:
they updated the preconfigured image, now it is working, you have to modify the config.txt before flashing it on raspberry pi, in the last rows there are a list of lcds display, you have to decomment the right one, then flash the image, for me the image was read-only so i had to figure out how to edit, here is official image www.waveshare.com/wiki/3.5inch_RPi_LCD_(C)#Image