waveshareteam/LCD-show #65: it doesn't work on new raspberry pi OS CodeName: BookWorm

descuidado, 2024-02-09:
it doesn't work on new raspberry pi OS CodeName: BookWorm aarch64
this repo needs an update.

vchrizz, 2024-06-27:
It still works fine, since Raspbian bookworm the files config.txt and cmdline.txt moved from `/boot/` to `/boot/firmware/` .
So I did following **before** I run ./LCD-show installation:
```
mv /boot/config.txt /boot/config.txt.old
mv /boot/cmdline.txt /boot/cmdline.txt.old
ln -s /boot/firmware/config.txt /boot/config.txt
ln -s /boot/firmware/cmdline.txt /boot/cmdline.txt
```

DennisRauscher, 2024-12-26:
After **a lot** of trial and error I made it work on bookworm x64 rp5, here is what I did:

1. fresh install of raspberry pi bookworm via official imager to sd card
2. follow the Guide https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)_Manual_Configuration "For Bookworm System" section, but with a change:
3. dont add the `99-fbturbo.~`, check if there is any configuration existing (from earlier trials) in the folder `/usr/share/X11/xorg.conf.d` and remove them (seems to be a issue with loading fbturbo (https://github.com/ssvb/xf86-video-fbturbo)
4. run startx, it should work immediatly
5. follow the rest of the guide

startup takes 2-5m for me from power to desktop.

Troubleshooting:
connect via ssh and run `startx`, check the error and the error log

Hope this helps!

ref: https://github.com/raspberrypi/bookworm-feedback/issues/237

littlefrank90, 2025-04-01:
None of the fixes above worked.
They keep selling these displays but none of them work because they haven't updated the software in 6 years. It's horrible to see. Not buying a Waveshare product ever again.